Qengineering/Jetson-Nano-Ubuntu-20-image #99: Broke Image Links

BeratTezer, 2024-09-24:
Images files are not able to download from given resources. I need them as soons as possible. If somehow I can get them, I will make them reachable for everyone.

BeratTezer, 2024-09-24:
I tried Waybak Machine snapshots but them also empty.

BeratTezer, 2024-09-24:
This problem might be related with my locations, I will aslo try with VPN.

Qengineering, 2024-09-24:
Indeed, location is the problem here. I've just tried to download the image and there is no problem whatsoever.
I will place a copy on Gdrive. Hopefully you can download it. It will take another hour before the upload of 9 GB is completed.....

Qengineering, 2024-09-24:
Please find a copy at [GDrive](https://drive.google.com/file/d/1L2H_sQC_kSILrcJteWg7htKxJirtDsZ9/view?usp=sharing).

BeratTezer, 2024-09-24:
Appreciate it. Thank you for your efforts 💖

closed 2024-09-24 by BeratTezer