Qengineering/Jetson-Nano-Ubuntu-20-image #102: camera Raspberry Pi v2 8MP (Sony IMX219)

Rodolfoloc, 2024-11-14:
Hello! I am using the image from this repository on the Jetson Nano. I have an application that requires a camera. I am using a Raspberry Pi v2 camera 8MP (Sony IMX219 sensor). It works fine with Python2, but nothing works with Python3. The GStreamer is disabled in Python3. Could you assist me with this? Thank you!

Qengineering, 2024-11-15:
Can you give me some more info? How did you use the camera with Python2? With OpenCV? Do you have a _small_ code snippet? How do you know that's disabled int Python3?

Rodolfoloc, 2024-11-15:
Can you give me some more info? How did you use the camera with Python2? With OpenCV? Do you have a small code snippet?
Response: Used how in https://github.com/jetsonhacksnano/CSI-Camera, file "simple_camera.py".

How do you know that's disabled int Python3?
Response: Running the command print(cv2.getBuildInformation()) is displays:
avresample: NO
GStreamer: NO
libv4l/v4l2: NO

Qengineering, 2024-11-16:
I'm afraid you're OpenCV isn't healthy anymore.
When you downloaded the image, the correct version of OpenCV can be found on the image.
There is no need to install OpenCV afterwards. When you do, chances are that you corrupt the installation.
Here is a screen dump of the original getBuildInformation().
[image]

Please check you're installation. If you have only one or two projects on the Nano, consider re-flashing.

Rodolfoloc, 2024-11-18:
Re-flashing solved the problem. Thanks!