Qengineering/Jetson-Nano-Ubuntu-20-image #104: Help to install clang 8 on ubuntu 22.04

abishekmuthian, 2024-12-21:
Greetings Qengineering team,

Your efforts in writing tools and documentation for Jetson Nano is highly appreciated.

While trying to upgrade to Ubuntu 20.04 from 18.04 using your guide, I accidentally upgraded to 22.04 and so now I'm trying to get  the tools as software from the ubuntu 20.04 working in 22.04.

I was able to install `gcc-8` using archive repo, but installing `clang-8` has been very difficult as none of the repositories work.

Kindly share your expertise in getting clang-8 working with Ubuntu 22.04.

Qengineering, 2024-12-21:
I don't recommend Ubuntu 22.04 as it doesn't support the CUDA environment. 
It looks as if everything is working, but as soon as some app is using CUDA, you might get unexpected behavior. 
That's why we never published an 22.04 version.

abishekmuthian, 2024-12-21:
Thank you for the quick reply, even if I manage to install cuda10.2 on 22.04 would it create issues?

Qengineering, 2024-12-21:
Yes. You are facing low-level driver issues.

abishekmuthian, 2024-12-21:
Thank you, I will use your 20.04 image and do a fresh install.

closed 2024-12-21 by abishekmuthian