Qengineering/Jetson-Nano-Ubuntu-20-image #109: Image is not flashing

Splendero, 2025-01-20:
# Please help I'm just a student

I have been flashing a 512gb micro sd, with the JetsonNanoUb20_3b.img.xz.

Here is my check sum: d738f1fe20088a1bdbd10e2358b512f7, which I assume that if my check sum is the same then my image is not corrupt.

I'm on Mac, but have also been trying to flash with a windows machine while following the guidelines but continue to have this problem. 

I have been downloading the image to my computer then flashing through a dongle onto the micro sd.

[image]

If there is anything that I'm not doing right now please let me know.

Qengineering, 2025-01-20:
You cannot use a USB dongle to flash an SD card.
The dongle only let you read/write files. Flashing with Balena requires another technique, not supported by an USB dongle.
You have to find a machine with a genuine SD port.

Splendero, 2025-01-20:
Thank you for that but I just tried with a genuine sd port. 

I followed this guide which had previously worked for a friend, 
Check Device Manager. See if something pops up under disk drives when you take the SD card in and out. If it’s still not there, restart PC with SD card in.
Then, open Disk Manager. You should see some disk, the size of your SD card, with a bunch of partitions. Be careful to only change partitions for the disk that is labelled “removable” or you will ruin your PC. Now, delete every partition in the SD card disk using the Disk Manager. If you delete everything on your computer because you delete your disk rather than the SD card, this is not my fault pls dont yell at me.
Now open cmd as an administrator. Run diskpart. Then run list disk. You should see your SD disk and the base PC disk. Run select disk <sd_disk> (ex. select disk 1). Then run list volume. You should see a volume with status “unusable” at the end of the list. Run select volume <unusable>. Then run list partition. There should be none. Run create partition primary. 

I followed the step on the Jetson setup guide on the [nvidia](https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write) website.

And still I get the same error

Qengineering, 2025-01-21:
I just tried, but it is possible to flash the SD card with an USB dongle. 

[image]

[image]

[image]

Assuming your file is not corrupted (give the correct checksum), you could try to use the [Raspberry Pi imager](https://www.raspberrypi.org/software/).
Sometimes it works better than Balena.
See issue #101

Splendero, 2025-01-22:
I have been checking and it still refuses to work, I have tried different sd cards and I'm just following the instructions on the start up Jetson guide for Mac. Could you try and download the image to see if you have problems. I'm downloading this [image](https://ln5.sync.com/dl/403a73c60/bqppm39m-mh4qippt-u5mhyyfi-nnma8c4t/view/default/14418794280004). Also if you have any alternative flashing options as far as other images would appreciate, I'm doing this for a school design time and I'm very short on time.

Qengineering, 2025-01-23:
I don't understand your steps quite well.
You have to download the image, flash it on a SD card. Insert it in your Jetson Nano. That's all.
Don't use a partition manager to alter all the partitions. You need them all!
First it thought it is your dongle whose causing you the trouble. However, after trying myself, it isn't.
As seen above, you can flash a SD card via a dongle with Balena. That's all you need to do.

Vari2ty331, 2025-01-29:
It should sound stupid to mention, but I also had an issue where the image would not flash even after proper settings with the image and Etcher with a big enough SD card. How I solved it was to mount the image with a Windows computer. Any image flashed with a linux never booted, but as I worked with a Windows 11 computer, the flash somehow worked. Same file, same SD card and USB adapter, same Etcher program, but just with Windows. I guess it might be a problem with Etcher on different OS? I cannot guess, but with windows, I managed to boot the SD card.

Qengineering, 2025-01-29:
@Vari2ty331,

Thank you! 
It is valuable information which helps others!

Splendero, 2025-09-20:
I found that using a dongle was the problem, I just wouldn't use a mac to flash again unless it had a sd card port

closed 2025-09-20 by Splendero