Qengineering/Jetson-Nano-Ubuntu-20-image #110: Not able to do first boot

sarmadsohaib, 2025-01-31:
Followed following step but not able to do the first boot with SD card. The screen gets black with white blinking underscore on the top left corner,

1. Formatted SD card with SD Card Formatter. Did not use the quick format option
2. Prepared SD card with RPi Imager and also with balenaEtcher. Same response with both
3. Inserted the SD card in Jetson Nano 
4. See few thing. Last message "Failed to start nvpmodel service." then the screen turns black with blinking white underscore on top left corner.

Please guide.

[image]

Qengineering, 2025-01-31:
Please check the other issues.
For instance the last one of https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/109

sarmadsohaib, 2025-01-31:
I have read various issues and also the one #109  suggested by you. But no luck. I am flashing the SD on windows as mentioned in #109 by someone. By the way I was able to connect the Jetson Nano serially using micro usb and putty. It worked. But as soon as I connect it with display and external power supply (5V, 4A) it does not. I am using VGA to HDMI converter as my display is VGA. 

On a side note I am able to successfully use Jetpack 4.4 with all the same hardware. Kindly suggest me some solutions to use your setups files. Thank you

Qengineering, 2025-01-31:
I see.
Given the fact you can run JetPack 4.4, I would manually upgrade the OS. This will take less time than try to solve this annoying issue.
You find the instructions here: https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html

Qengineering, 2025-01-31:
Why using JetPack 4.4 while we currently have 4.6?
All your errors may be caused by mismatching software/firmware and incomplete system updates.
Please use Perplexity AI. When asked, it came with valuable advises.

sarmadsohaib, 2025-01-31:
You are right about 4.6. I am not able to boot my Jetson Nano with any versions of 4.6. I have tried really hard to find a reason but no luck.

yuvanshankar-ys, 2025-04-27:
Hi, could u update have u got the solution for this? Even i tried jetpack 4.6.x, i couldnt resolve the issue.
Thanks

sarmadsohaib, 2025-04-27:
Hi, I am using JetPack 4.4 on Jetson Nano now if that is what you are asking.

yuvanshankar-ys, 2025-04-28:
Yeah, Is it working with the externel monitor for u?