Qengineering/Jetson-Nano-Ubuntu-20-image #112: Partition does not have a FAT file system

Peterlearnhow2code, 2025-02-11:
I'm using a microSD card 64GB, with the JetsonNanoUb20_3b.img.xz. I connected microSD card with my Lexar microSD Card USB 3.2 Reader and flashed the xz image directly using Imager after It verified I received this error 

[image]

Please help

Qengineering, 2025-02-11:
If you use a dongle: 
Once the image is flashed, BalenaEtcher / Rpi imager can not valided the newly burned image. It is well possible that the image is written perfectly. Please check.

Peterlearnhow2code, 2025-02-11:
when I insert the SD card in to my Jetson Nano and it just shows up this screen for about an hour
[image]
should I replug the power jack?

Qengineering, 2025-02-11:
Oh dear, that's not good. You could try rebooting by unplugging the power. No guarantees.

Peterlearnhow2code, 2025-02-11:
I know what the problem is, I purchased a cheap Sandisk SD with 100MB/s, so that Jetson does not handle it and can not boot up. 
Thank you so much for your fast response

closed 2025-02-11 by Peterlearnhow2code