Qengineering/Jetson-Nano-Ubuntu-20-image #115: Stuck at Nvidia Logo on the boot up for jetson nano 2GB.

abdosalem490, 2025-04-12:
I am using Jetson Nano 2GB version I flash this image into an SD card formatted as exFAT. However the kernel doesn't boot up and just shows Nvidia Logo on the screen. these are the logs:
```
Welcome to minicom 2.9

OPTIONS: I18n 
Port /dev/ttyACM1, 20:16:32

Press CTRL-A Z for help on special keys

[0000.125] [L4T TegraBoot] (version 00.00.2018.01-l4t-80a468da)
[0000.131] Processing in cold boot mode Bootloader 2
[0000.135] A02 Bootrom Patch rev = 1023
[0000.139] Power-up reason: pmc por
[0000.142] No Battery Present
[0000.145] pmic max77620 reset reason
[0000.148] pmic max77620 NVERC : 0x40
[0000.151] RamCode = 1
[0000.154] Platform has DDR4 type RAM
[0000.157] max77620 disabling SD1 Remote Sense
[0000.161] Setting DDR voltage to 1125mv
[0000.165] Serial Number of Pmic Max77663: 0x2316ed
[0000.173] Entering ramdump check
[0000.176] Get RamDumpCarveOut = 0x0
[0000.179] RamDumpCarveOut=0x0,  RamDumperFlag=0xe59ff3f8
[0000.184] Last reboot was clean, booting normally!
[0000.189] Sdram initialization is successful 
[0000.193] SecureOs Carveout Base=0x00000000ff800000 Size=0x00800000
[0000.199] Lp0 Carveout Base=0x00000000ff780000 Size=0x00001000
[0000.205] BpmpFw Carveout Base=0x00000000ff700000 Size=0x00080000
[0000.211] GSC1 Carveout Base=0x00000000ff600000 Size=0x00100000
[0000.216] GSC2 Carveout Base=0x00000000ff500000 Size=0x00100000
[0000.222] GSC4 Carveout Base=0x00000000ff400000 Size=0x00100000
[0000.228] GSC5 Carveout Base=0x00000000ff300000 Size=0x00100000
[0000.234] GSC3 Carveout Base=0x00000000bf300000 Size=0x00d00000
[0000.250] RamDump Carveout Base=0x00000000ff280000 Size=0x00080000
[0000.256] Platform-DebugCarveout: 0
[0000.259] Nck Carveout Base=0x00000000ff080000 Size=0x00200000
[0000.265] Non secure mode, and RB not enabled.
[0000.270] Read GPT from (4:0)
[0000.998] Csd NumOfBlocks=262146048
[0001.003] Set High speed to 1
[0001.011] Using GPT Primary to query partitions
[0001.016] Loading Tboot-CPU binary
[0001.027] Verifying TBC in OdmNonSecureSBK mode
[0001.037] Bootloader load address is 0xa0000000, entry address is 0xa0000258
[0001.044] Bootloader downloaded successfully.
[0001.048] Downloaded Tboot-CPU binary to 0xa0000258
[0001.053] MAX77620_GPIO5 configured
[0001.056] CPU power rail is up
[0001.059] CPU clock enabled
[0001.063] Performing RAM repair
[0001.066] Updating A64 Warmreset Address to 0xa00002e9
[0001.071] Read GPT from (4:0)
[0001.079] Using GPT Primary to query partitions
[0001.083] Loading NvTbootBootloaderDTB
[0001.113] Verifying NvTbootBootloaderDTB in OdmNonSecureSBK mode
[0001.184] Bootloader DTB Load Address: 0x83000000
[0001.189] Read GPT from (4:0)
[0001.197] Using GPT Primary to query partitions
[0001.201] Loading NvTbootKernelDTB
[0001.230] Verifying NvTbootKernelDTB in OdmNonSecureSBK mode
[0001.302] Kernel DTB Load Address: 0x83100000
[0001.306] Read GPT from (4:0)
[0001.314] Using GPT Primary to query partitions
[0001.318] Loading cboot binary
[0001.355] Verifying EBT in OdmNonSecureSBK mode
[0001.396] Bootloader load address is 0x92c00000, entry address is 0x92c00258
[0001.403] Bootloader downloaded successfully.
[0001.407] Read GPT from (4:0)
[0001.415] Using GPT Primary to query partitions
[... 354 lines cut ...]
34746376 bytes read in 2019 ms (16.4 MiB/s)
append: tegraid=21.1.2.0.0 ddr_die=2048M@2048M section=256M memtype=0 vpr_resiz 
## Flattened Device Tree blob at 83100000
   Booting using the fdt blob at 0x83100000
ERROR: reserving fdt memory region failed (addr=0 size=0)
ERROR: reserving fdt memory region failed (addr=0 size=0)
   Using Device Tree in place at 0000000083100000, end 000000008317ced3
copying carveout for /host1x@50000000/dc@54200000...
copying carveout for /host1x@50000000/dc@54240000...

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.253 (root@nano) (gcc version 7.5.0 (Ubuntu/Lin2
[    0.000000] Boot CPU: AArch64 Processor [411fd071]
[    0.000000] OF: fdt:memory scan node memory@80000000, reg size 32,
[    0.000000] OF: fdt: - 80000000 ,  3f200000
[    0.000000] OF: fdt: - c0000000 ,  3ee00000
[    0.000000] Found tegra_fbmem: 00800000@92cb4000
[    0.000000] earlycon: uart8250 at MMIO32 0x0000000070006000 (options '')
[    0.000000] bootconsole [uart8250] enabled

```

and nothing is printed after `bootconsole [uart8250] enabled` as per [issue#2](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/2), the image wasn't made for the jetson nano 2GB but as mentioned in [issue#94](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/94), it's working. so I can't figure out what is the problem.

Qengineering, 2025-04-13:
You have to upgrade your original Ubuntu 18.04 to 20.04 manually.
Follow [this guide](https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html) carefully.

Qengineering, 2026-03-14:
The Jetson Nano (**not the Orin**), ships with an Ubuntu 18.04 OS.
If you want to upgrade to Ubuntu 20.04, you are facing a challenge.
We have made it easy for you, by providing a working version of Ubuntu 20.04 on a SD image.
Now, you don't have to manually upgrade your original OS yourself.
BTW, Ubuntu 20.04 ships with Python 3.8, needed, for instance, with ROS

ThisisBillhe, 2026-03-15:
I am using Jetson Nano B01 4 GB. However, I got stuck at the Nvidia logo...The tf card was flashed using balenaEtcher. Any idea what the reason might be?